Open-Cascade-SAS/OCCT #100: Documentation migration to GitHub

dpasukhi, 2024-10-12:
The main idea is simplify access for documentation for community.
The dox folder will have a copy in Wiki. The maintaining logic about 2 overview is not defined. Probably both will be supported or another way will be chosen. 

- [ ] Create a action to build refman documentation
- [ ] Migrate Overview documentation to pure doxygen and HTML syntax
- [ ] Create a Wiki for OCCT
- [ ] Create a GitHub self-hosted refman documentation

dpasukhi, 2024-10-17:
Wiki is created:https://github.com/Open-Cascade-SAS/OCCT/wiki
I was copied from my own https://github.com/dpasukhi/OCCT/wiki

dpasukhi, 2024-10-19:
Self hosted will be not used.
https://dev.opencascade.org/doc/refman/html/
will be updated periodically.

dpasukhi, 2024-10-20:
Merged weekly building refman and Wiki is open

closed 2024-10-20 by dpasukhi

benj5378, 2024-11-02:
@dpasukhi, I want to express great gratitude for your efforts migrating to GitHub. It will make it much easier to contribute with PRs and give comments on issues, now that the code is on a widely known platform. Thank you very much for your great efforts migrating here.