Open-Cascade-SAS/OCCT #115: About TKOpenGL, When I create a View, the following error occurs

WY-Huang, 2024-10-23:
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Error: glBufferData (GL_ARRAY_BUFFER,4056,0x284B9880000) Id: 2 failed with GL_INVALID_OPERATION
TKOpenGl | Type: Performance | ID: 0 | Severity: Low | Message:
  VBO creation for Primitive Array has failed for 169 vertices. Out of memory?
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Error: 2D texture 228x604 IF: GL_SRGB8_ALPHA8 PF: GL_RGBA DT: GL_UNSIGNED_BYTE can not be created with error GL_INVALID_OPERATION [fbo0_main:color]
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Error! Main FBO 228x604@0 initialization has failed

dpasukhi, 2024-10-23:
I will move the issue to discussion.
@WY-Huang Don't you mind.

Additionally, could you share the OS and available hardware?

WY-Huang, 2024-10-23:
Not at all!
Thank you! The issue has been resolved.

dpasukhi, 2024-10-23:
@WY-Huang great! Could you share your fix or just way to investigate the cause?