Open-Cascade-SAS/OCCT #172: Fillet can not round a face

farahats9, 2024-11-29:
### Category

Modeling

### Severity

Major

### Description

I know this bug is in the original tracker [25478](https://tracker.dev.opencascade.org/view.php?id=25478)
but it has been 10 years since it was reported and there is no sign of any attention or progress done on this. Please someone needs to look into it since it is feels like the most important problem now in FreeCAD

### Expected Behavior

if the fillet radius is big enough that there are no linear faces remaining, it should round the face.

### Actual Behavior

The fillet will work as long as the radius is small enough to keep some face remaining. if it is big it will not work.

### Sample Code or DRAW Tcl Script

see original issue for more description and examples. [25478](https://tracker.dev.opencascade.org/view.php?id=25478)

### Operating System

Linux

### Compiler

GCC

### Bitness

64-bit

### OCCT Version

latest

GeneGH, 2024-11-29:
What do you mean by "round the face"? 

* Use the requested radius and leave sharp corners?

* Change the radius to fit?

* Just create some artibrary shape that looks "round"?

farahats9, 2024-11-29:
It may be easier to explain visually:

I have the following simple shape with height 10 mm
[image]

now I want to apply a fillet on all sides that is 4 mm, will work without any problem
[image]

because there is still a face connecting the 2 red fillets there is no problem so far.

now if I want to make the fillet 5mm that means they will touch and the grey face between them will have 0 mm length and will refuse to do the fillet with the error:
```BRep_API: command not done```

I hope this makes the issue a bit clearer, this is also in **chamfer** not just a fillet problem, let me know if you need any information or if I can be of any help.

GeneGH, 2024-11-30:
I understand the issue. I have seen it many times.

The question is what should happen when the fillet cannot be completed as specified.

GeneGH, 2024-11-30:
OK, now they are tangent at the former face along the edge. What happens at the other ends? Sharp break in the curve? Changed radius of the fillet so that the other ends can be tangent to those faces?

farahats9, 2024-11-30:
In my example both the top and bottom fillets have the same radius so both of them would be tangent without changing any radius.

But if I understand you correctly you are asking what if they cannot be tangent due to radius difference or other constraints, from my experience in other CAD software, you should never change the other fillet's radius, keep the sharp corners and let the dimension changes in the user's control.

dragazo, 2024-12-11:
I don't know if this would help in the implementation, but I believe the behavior used by commercial CAD software is equivalent to performing the fillet on each edge loop separately and then taking the intersection. So in this example we would duplicate the part, perform the top fillet on one copy, perform the bottom fillet on the other copy, and then do a boolean intersection to get the final result. As long as boolean intersections don't have some other issue, this should work for both over-fillet and critical-fillet (as is used in this issue), so two birds with one stone.

angelobartolome, 2025-08-26:
I've manually tested @dragazo's suggestion on a simpler version, filleting two edges until they overlap, seems quite promising

[image]

drwho495, 2025-08-27:
does this have the Modified and Generated methods? freecad has to have those for its toponaming algorithm

dpasukhi, 2026-05-16:
> [@dpasukhi](https://github.com/dpasukhi) what are your thoughts on [#172 (comment)](https://github.com/Open-Cascade-SAS/OCCT/issues/172#issuecomment-4418443364) ?

We discuss a little in Discord. That comment is helpful to know solutions and possible problems.
We will not directly implement original algorithm, but we will consolidate the logic around.

FloFaber, 2026-05-20:
Fully agree with @MisterMakerNL. This is the No.1 reason for me sticking with other CAD software instead of using FreeCad.

MisterMakerNL, 2026-06-05:
If we can fix that, 

> > > From our side, I can't provide resources to solve the issues in that fast way. Unfortunately, at the moment, we do not have clients who are interested in that type of problem.
> > 
> > 
> > This is why FreeCAD is never going to get mainstream. It's a basic thing that is common in every other CAD program, and OCCT isn't interested in developing a CAD kernel for a CAD application. Staying with OCCT is never going to work for FreeCAD.
> 
> You always can develop your own kernel ;) Or you can sponsorship the development by request the commercial bug fix ;)
> 
> Or, which is more welcoming - prepare a patch and create PR with deep rework ;)
> 
> OCCT never restrict no one to use another kernels. You are welcome to use another open source kernel. Or you can use commercial kernel and pay full price ;)

A billion dollar-company asking for sponsorship from a Opensource organization.... Come on man that's bs and you know it.

Your org. could even sponsor Freecad get a tax refund, FreeCAD pays you money back = free money. Although I think FPA didn't registered as a charity 😭.

Gladly we got some awesome people in the community @angelobartolome ! Hope it'll work!