Open-Cascade-SAS/OCCT #212: It's not a Bug, a compile error.

fynkxo, 2024-12-27:
### Description

Third-party library freetype was changed "char* --> unsigned char*" in struct FT_Outline_ from VER-2-13-3， See:[VER-2-13-3](https://github.com/freetype/freetype/releases/tag/VER-2-13-3) . This change caused an occt compilation error in StdPrs_BRepFont.cxx:460.

### Expected Behavior

Never.

### Actual Behavior

Never.

### Sample Code or DRAW Tcl Script

Never.

### Operating System

Windows

### Compiler

MSVC

### Bitness

64-bit

### OCCT Version

7.8

dpasukhi, 2024-12-27:
Dear @fynkxo 
Thank you for the bug report.
I'm glad to inform that bug is fixed. You can cherry pick on latest release the changes or use latest master.
Each master passed all of our internal and external tests.
The fix: https://github.com/Open-Cascade-SAS/OCCT/pull/49

You are welcome to share more issues, if you find :)

Best regards,
Dmitrii.

dpasukhi, 2024-12-27:
Original bug report: https://github.com/Open-Cascade-SAS/OCCT/discussions/48

dpasukhi, 2024-12-27:
The commit : 7236e83dcc1e7284e66dc61e612154617ef715d6

closed 2024-12-27 by dpasukhi

fynkxo, 2024-12-28:
Thank you for your answer.