Open-Cascade-SAS/OCCT #230: AIS_Manipulator issues

pycode74, 2025-01-02:
### Description

I am trying to use the AIS_Manipulator to select and perform translations of a box (created with BRepPrimAPI_MakeBox).

The translation works fine the first time, but somethings seems to break after that.

### Expected Behavior

- I click on the box to select it
- The manipulator appears
- I click and drag one of the manipulators arrows to translate the box
- I release the mouse
- I can click on the box to select it again
- The manipulator appears
- I click and drag one of the manipulators arrows to translate the box

### Actual Behavior

- I click on the box to select it
- The manipulator appears
- I click and drag one of the manipulators arrows to translate the box (works fine)
- I release the mouse
- I cannot re select the box again

### Sample Code or DRAW Tcl Script

```
void ViewerWidget::init()
{
    // ...
    gp_Ax2 anAxisManipulator;
    myManipulator = new AIS_Manipulator(anAxisManipulator);
    myManipulator->SetPart(AIS_ManipulatorMode::AIS_MM_Translation, Standard_True);
    myManipulator->SetPart(AIS_ManipulatorMode::AIS_MM_Rotation, Standard_False);
    myManipulator->SetPart(AIS_ManipulatorMode::AIS_MM_Scaling, Standard_False);
    myManipulator->EnableMode(AIS_ManipulatorMode::AIS_MM_Translation);
    // ...
}

void ViewerWidget::mousePressEvent(QMouseEvent* event)
{
    int x = event->pos().x();
    int y = event->pos().y();

    myContext->MoveTo(x, y, myView, Standard_True);

    if (event->button() == Qt::LeftButton) {
        myContext->SelectDetected();
        Handle(AIS_InteractiveObject) selected = myContext->FirstSelectedObject();
        if (selected) {

            myManipulator->Attach(selected);
            myContext->Display(myManipulator, Standard_True);

            if (myManipulator->HasActiveMode()) {
                myManipulator->StartTransform(x, y, myView);
            }
        }
        else {
            if (myContext->IsDisplayed(myManipulator)) {
                myContext->Erase(myManipulator, Standard_True);
            }
        }
    }
}

void ViewerWidget::mouseMoveEvent(QMouseEvent* event)
{
    int x = event->pos().x();
    int y = event->pos().y();

    myContext->MoveTo(x, y, myView, Standard_True);

    if (myManipulator->HasActiveMode()) {
        myManipulator->Transform(x, y, myView);
        myView->Redraw();
    }
}

void ViewerWidget::mouseReleaseEvent(QMouseEvent* event)
{
    int x = event->pos().x();
    int y = event->pos().y();

    myContext->MoveTo(x, y, myView, Standard_True);

    if (myManipulator->HasActiveMode()) {
        myManipulator->StopTransform();
        myManipulator->DeactivateCurrentMode();
        myManipulator->Detach();
        myView->Redraw();
    }
}
```

### Operating System

Windows

### Compiler

MSVC

### Bitness

64-bit

### OCCT Version

7.8

dpasukhi, 2025-01-03:
Hello, could you please check on latest version (master).

pycode74, 2025-01-03:
> Hello, could you please check on latest version (master).

This was using the latest master build (7.8.2).

pycode74, 2025-01-18:
Any update on this? Is it reproducible?

pycode74, 2025-01-28:
I have commented out this line, but the problem persists:

```
//myManipulator->DeactivateCurrentMode();
```

pycode74, 2025-01-29:
Thanks @mzernova, it works definitely better.
However, I am losing the object selection (and the highlight) as soon as I start using the manipulator.
Is there a way to fix that?