Open-Cascade-SAS/OCCT #231: glTF cannot import some files

oursland, 2025-01-04:
### Description

When importing some glTF files, an exception is thrown: https://github.com/Open-Cascade-SAS/OCCT/blob/master/src/TopoDS/TopoDS_Shape.hxx#L94

### Expected Behavior

OCCT imports the glTF file.

### Actual Behavior

An exception is thrown stating `Moving with scaling transformation is forbidden`.

### Sample Code or DRAW Tcl Script

See file.

### Operating System

macOS

### Compiler

Clang

### Bitness

64-bit

### OCCT Version

7.8

### Additional Files

[example.glb.zip](https://github.com/user-attachments/files/18305822/example.glb.zip)

dpasukhi, 2025-01-04:
@oursland OCCT have no issue to read that file.
```
Draw[9]> ReadGltf D1 c:/example.glb
glTF reader, scene node '1' defines unsupported scaling 0.993799 0.993799 1
Mesh c:/example.glb
[13276 nodes] [4782 2d elements]
[read in 0.0541816 s]
```
The exception probably on the application side when it try to work with scaled shapes.
Could you share more details.

oursland, 2025-01-05:
I'm encountering this from within FreeCAD: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Import/App/ReaderGltf.cpp#L83

dpasukhi, 2025-01-05:
Oke. Then I will revert exception integrated in version 7.6.
Indeed decision with exception is not completed. Need to develop something with clear idea and description what to do.
I'll think about final approach.
But for now I will disable exception

dpasukhi, 2025-01-05:
Exception disabled in latest master.
Marked as done.

closed 2025-01-05 by dpasukhi