Open-Cascade-SAS/OCCT #245: Boolean cut has issues when it involves a common Edge between argument and the tool.

kavindaTurbogen, 2025-01-06:
### Description

In this operation the argument and the tool both have a common edge (Attachment: Highlighted in blue) which affects the boolean cut operation.
When we rotate or offset the shape to avoid the common edge boolean cut operation works as intended.
BRep files are attached in the opencascade forum issue (link included below).

### Expected Behavior

Boolean cut should be done without issues even a common edge exists.

### Actual Behavior

Boolean operation is not completed

### Sample Code or DRAW Tcl Script

  BRepAlgoAPI_Cut _operation;

void geo_OperationCut::setArguments(const TopTools_ListOfShape& targetList, const TopTools_ListOfShape& toolList)
{
	_operation.SetArguments(targetList);
	_operation.SetTools(toolList);
}

void geo_OperationCut::build()
{
	if (validate(_operation.Arguments(), _operation.Tools()))
		try
	{
		OCC_CATCH_SIGNALS
			_operation.Build();
	}
	catch (const std::exception&)
	{
		return;
	}
}

### Operating System

Windows

### Compiler

MSVC

### Bitness

64-bit

### OCCT Version

7.8

### Additional Files

[https://dev.opencascade.org/content/boolean-cut-has-issues-when-it-involves-common-edge-between-argument-and-tool] (
[image]
url)

CuiYouWei888, 2025-04-22:
After a brief analysis, I found a bug in the intersection between revolution surface and sphere surface at the sphere's singular points. The issue can be resolved by switching to parametric surface-surface intersection method.

kavindaTurbogen, 2025-04-22:
Thanks for the update. I will try this solution and let you know the results.