Open-Cascade-SAS/OCCT #334: Opencascade Conan Package: Version conflict

trns1997, 2025-02-03:
### Description

Please refer to the following issue on conan io for details : https://github.com/conan-io/conan-center-index/issues/26515.
Opencascade no longer build with conan after updates made in the recipe of fontconfig. Specifically the following commit: https://github.com/conan-io/conan-center-index/commit/201d8e3ee0d1dca10c89450ae5a72da2dbbdace6

### Expected Behavior

Successful install of opencascade as a conan package

### Actual Behavior

ERROR: Version conflict: fontconfig/2.13.93->freetype/2.13.2, opencascade/7.6.2->freetype/2.13.0.

### Sample Code or DRAW Tcl Script

Refer to https://github.com/conan-io/conan-center-index/issues/26515

### Operating System

Linux

### Compiler

GCC

### Bitness

64-bit

### OCCT Version

7.5

dpasukhi, 2025-02-03:
@trns1997 Hello.
Thank you for information. I can help to migrate conan to use occt 7.9 after release.
Do you mind to use 7.9?

dpasukhi, 2025-02-03:
Oh, I see that it is already resolved.
And see that there are a few PR to move to latest version

trns1997, 2025-02-03:
Yes I just wanted to keep the open cascade repo in the loop. In case the problem was complicated to resolve.

trns1997, 2025-02-04:
Issue resolved thanks to the conan maintainers refer to: https://github.com/conan-io/conan-center-index/pull/26518

closed 2025-02-04 by trns1997