Open-Cascade-SAS/OCCT #353: Collecting related OCCT products on GitHub

dpasukhi, 2025-02-09:
### Description

Needs to collect GitHub open source solutions related to OCCT.
It can be some list (example, https://dev.opencascade.org/about/projects_and_products) but in GitHub.
The main idea - share more advanced samples with description to community.

### Use Case

Search for samples and ideas

### Benefits

Having more samples and projects in view.
Popularity for open source solutions.

dpasukhi, 2025-02-11:
If someone knows some interesting OpenSource projects and share here, it will be helpful

ii8, 2025-02-23:
[Horizon EDA](https://horizon-eda.org/) and [Dune 3D](https://dune3d.org/) both use OCCT.

oursland, 2025-02-23:
It's already on the aforementioned list, but [FreeCAD](https://github.com/FreeCAD/FreeCAD) is on GitHub and uses OCCT.

conorbergin, 2025-03-21:
[Build123d](https://github.com/gumyr/build123d) is a popular python dsl for modelling.
[Gmsh](https://gmsh.info/) and [Netgen](https://gmsh.info/) are two of the most popular mesh generation libraries and they both have an option to use opencascade for modelling.

zerger, 2025-06-19:
[mayo](https://github.com/fougue/mayo) 3D CAD viewer and converter based on Qt + OpenCascade

dpasukhi, 2025-12-27:
Thank you for the sample, in January I will open form to collect all open source projects and put some info into "samples"folder of the repo with some sorting and organisation by area type and so on.
It means, sample folder became folder of references to opensource products which use OCCT.
I will publish more info here and on announcements when it will started.

gsdali, 2026-04-14:
We've been working on a swift wrapper  for OCCT targeting 8.0.0 - [OCCTSwift](https://github.com/gsdali/OCCTSwift)

This is the basis for a number of other projects I'm working on targeting MacOS/iOS. It's also been a way for me to learn agentic coding. It's pretty much a just a wrapper, but it does translate to Idiomatic Swift. The latest release has added features to enable better integration with AI/ML which have been enabled by BREPGraph.

Along the way We've  also produced

- [OCCTSwiftViewport](https://github.com/gsdali/OCCTSwiftViewport) - A Metal based viewport and rendering engine. It also has a MacOC/IOS app that demos a lot of the OCCT features and has been my test harness through the project
- [OCCTSwiftScripts](https://github.com/gsdali/OCCTSwiftScripts) - Inspired by Cadquery and OpenSCAD it is  swift style scripting harness to build models in OCCT
- [OCCTMCP](https://github.com/gsdali/OCCTMCP) - a MCP Server to allow agentic tools to interact with OCCTSwift