Open-Cascade-SAS/OCCT #367: FreeCAD 1.0 load one iges file result Access violation but not reproduce on FreeCAD 0.21.2

lleheaven, 2025-02-13:
### Description

[52mm Pressure Gauge.zip](https://github.com/user-attachments/files/18780414/52mm.Pressure.Gauge.zip)

### Expected Behavior

Could load this iges file like old version FreeCAD

### Actual Behavior

failed. 
16:23:37  <Exception> Access violation
16:23:37  pyException: Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\freecad\module_io.py", line 16, in OpenInsertObject
    getattr(importerModule, importMethod)(objectPath, *importArgs, **importKwargs)
<class 'RuntimeError'>: FreeCAD exception thrown (Access violation)

### Sample Code or DRAW Tcl Script

FreeCAD 1.0

### Operating System

Windows

### Compiler

GCC

### Bitness

64-bit

### OCCT Version

7.8

dpasukhi, 2025-02-13:
Dear @lleheaven.
OCCT team do not informed about FreeCAD functionality. Please avoid internal samples. For sure when it is only single way of sample - then ok. But better to avoid any python or FreeCAD version samples

dpasukhi, 2025-02-13:
Can't reproduce:

[image]

dpasukhi, 2025-02-13:
7.8 lates delivery can't reproduce.
latest or 7.8 not reproducable

gkv311, 2025-02-13:
> 7.8 and latest delivery can't reproduce.
> latest or 7.8 not reproducable

If you want to find the place with NULL dereference - just catch exceptions in DRAW Harness under debugger and trace for Access Violations. If you want reproduce application environment that doesn't catches `SIGSEV` (e.g. doesn't call `OSD::SetSignal(false)` or it doesn't work as expected / overridden by another tracer from Python, etc.) - try `dsetsignal unset` in DRAW (and better in Release build).

For me, debugger stops at NULL de-reference at `BSplineRes->FirstParameter()` in `IGESToBRep_BasicCurve::TransferBSplineCurve()`.

dpasukhi, 2025-02-13:
I was attached by debugger, but thank you. I will check one more time.

dpasukhi, 2025-02-13:
Found, thank you.

[image]

lleheaven, 2025-02-13:
[iges_fail.zip](https://github.com/user-attachments/files/18785879/iges_fail.zip)
Add more sample files ~~

dpasukhi, 2025-02-13:
https://tracker.dev.opencascade.org/view.php?id=30731
Added validation for empty poles cases. (6 years ago).
For 7.9 I will add some additional checks to avoid crash. I was trying to create a degenerated line, but no benefits in the result. For now added short validation to degenerated BSpline.
https://github.com/Open-Cascade-SAS/OCCT/pull/368

(the poles (1) == (2) && (3) == (4)) and (1) distance to (4) = 1e-4.

dpasukhi, 2025-02-13:
@lleheaven all of them have the same issue. Will be fixed on 7.9
At the moment only removing of degenerated BSpline is prepared to avoid null dereferencsing. Probably in the future, possible to thing about create degenerated line.

closed 2025-02-17 by dpasukhi