Open-Cascade-SAS/OCCT #378: Revolved shape in STEP file is imported inverted

elias-bananaz, 2025-02-17:
### Description

The attached file is imported incorrectly. It appears as if the revolved shape is imported with the complementary angles instead.

### Expected Behavior

Shape should look like this:

[image]

### Actual Behavior

It looks like this instead:

[image]

### Sample Code or DRAW Tcl Script

```
pload XDE OCAF VISUALIZATION MODELING

# read STEP file into XDE document using STEPCAFControl_Reader
jsupload /models/occ/as1-oc-214.stp
ReadStep D as1-oc-214.stp

# display the document in 3D viewer using XCAFPrs_AISObject
vinit View1
XDisplay -dispMode 1 D -explore
vfit
```

(sample from https://draw.sview.ru/xde-import-step/)

### Operating System

Windows

### Compiler

MSVC

### Bitness

64-bit

### OCCT Version

latest

### Additional Files

STEP file to reproduce:
[b15.zip](https://github.com/user-attachments/files/18827408/b15.zip)

dpasukhi, 2025-02-17:
7.6.3 it was correct.
Regression. But unfortunately, can't be done for 7.9.0. Will be planned for the next release

dpasukhi, 2025-02-17:
Yes, if the fix will be ready, it will be available in dev

elias-bananaz, 2025-03-03:
I have another file that has a problem with a revolved surface. Not sure if it's the same problem, but I'm sharing here nonetheless.[Blades lock.zip](https://github.com/user-attachments/files/19053722/Blades.lock.zip)

You can see a sort of "blemish" at the center of the shape, when the surface is triangulated.
[image]

dpasukhi, 2025-03-04:
The regression commit: b0f92d3d5c78548a1e5d14f7b7c28463a3981092

dpasukhi, 2025-03-30:
@elias-bananaz the fix is planned to be done.
But case if fixed. Need more changes for general case

dpasukhi, 2025-09-04:
Original issue resolved. In case of the keeping the problem, reopen the issue or create a new one.

closed 2025-09-04 by dpasukhi

dpasukhi, 2025-09-06:
> > Please report it separately. It is new dateset. Please create a new bug report
> 
> Will do, thank you!

No needs for any info to reproduce,you can create a bug report with empty details and these dataset