Open-Cascade-SAS/OCCT #382: One surface from step file is wrongly imported or tesselated

LorienGL, 2025-02-19:
### Description

There is one surface (seems a cylinder) in this step file is not imported properly or the tessellation fails for it.

Attached the file, the image showing the faulty suface and and image of another cad showing the surface correctly.

Tried with CSharp_D3D example and with my own app, using STEPControl_Reader and STEPCAFControl_Reader, both with the same result.

Regards,

### Expected Behavior

Surface is shown as a cylinder like other CAD systems show it

### Actual Behavior

Surface is faulty (in graphic or topology, had not analyzed it as i dunno how yet).

### Sample Code or DRAW Tcl Script

-----> USING CSharp_D3D sample as is:  

bool ImportStep (const TCollection_AsciiString& theFileName)
  {
    STEPControl_Reader aReader;
    if (aReader.ReadFile (theFileName.ToCString()) != IFSelect_RetDone)
    {
      return false;
    }

    bool isFailsonly = false;
    aReader.PrintCheckLoad( isFailsonly, IFSelect_ItemsByEntity );

    int aNbRoot = aReader.NbRootsForTransfer();
    aReader.PrintCheckTransfer (isFailsonly, IFSelect_ItemsByEntity);
    for (Standard_Integer aRootIter = 1; aRootIter <= aNbRoot; ++aRootIter)
    {
      aReader.TransferRoot (aRootIter);
      int aNbShap = aReader.NbShapes();
      if (aNbShap > 0)
      {
        for (int aShapeIter = 1; aShapeIter <= aNbShap; ++aShapeIter)
        {
          myAISContext()->Display (new AIS_Shape (aReader.Shape (aShapeIter)), Standard_False);
        }
        myAISContext()->UpdateCurrentViewer();
      }
    }
    return true;
  }

### Operating System

Windows

### Compiler

MSVC

### Bitness

64-bit

### OCCT Version

latest

### Additional Files

[image]

[33.zip](https://github.com/user-attachments/files/18864252/33.zip)

[image]

LorienGL, 2025-02-19:
[image]

dpasukhi, 2025-02-19:
One of the cylinders are corrupted (missed) after reading and fixed. Probably issue on original file and then ShapeFix can't fix it. Need to check ShapeHealing.

[image]