Open-Cascade-SAS/OCCT #384: For some STEP files we fail for geometric tolerances to read the "geometric_tolerance.magnitude"

T-Dunker, 2025-02-19:
### Description

For some STEP files we fail for geometric tolerances to read the "geometric_tolerance.magnitude" as a StepBasic_MeasureWithUnit. The parameter 3 may be of type StepRepr_ReprItemAndLengthMeasureWithUnitAndQRI or StepRepr_ReprItemAndPlaneAngleMeasureWithUnitAndQRI.

There is a method
```
Handle<StepBasic_MeasureWithUnit> GetMeasureWithUnit () const 
```
inherited from StepRepr_ReprItemAndMeasureWithUnit (parent of the parent), which would return the needed handle, but none of the parent Types satisfies isKind("StepBasic_MeasureWithUnit")

There are multiple tolerances concerned in src\RWStepDimTol\:
  - RWStepDimTol_RWAngularityTolerance.cxx
  - RWStepDimTol_RWCircularRunoutTolerance.cxx
  - RWStepDimTol_RWCoaxialityTolerance.cxx
  - RWStepDimTol_RWConcentricityTolerance.cxx
  - RWStepDimTol_RWCylindricityTolerance.cxx
  - RWStepDimTol_RWFlatnessTolerance.cxx
  - RWStepDimTol_RWGeometricToleranceWithDatumReference.cxx
  - RWStepDimTol_RWGeometricToleranceWithDefinedAreaUnit.cxx
  - RWStepDimTol_RWGeometricToleranceWithDefinedUnit.cxx
  - RWStepDimTol_RWGeometricToleranceWithMaximumTolerance.cxx
  - RWStepDimTol_RWGeometricToleranceWithModifiers.cxx
  - RWStepDimTol_RWLineProfileTolerance.cxx
  - RWStepDimTol_RWModifiedGeometricTolerance.cxx
  - RWStepDimTol_RWParallelismTolerance.cxx
  - RWStepDimTol_RWPerpendicularityTolerance.cxx
  - RWStepDimTol_RWPositionTolerance.cxx
  - RWStepDimTol_RWRoundnessTolerance.cxx
  - RWStepDimTol_RWStraightnessTolerance.cxx
  - RWStepDimTol_RWSurfaceProfileTolerance.cxx
  - RWStepDimTol_RWSymmetryTolerance.cxx
  - RWStepDimTol_RWTotalRunoutTolerance.cxx
  - RWStepDimTol_RWUnequallyDisposedGeometricTolerance.cxx

Among the NIST PMI step file examples there are some, where the 3rd parameter of the geometric tolerances is of the form
```
#id=(
LENGTH_MEASURE_WITH_UNIT()
MEASURE_REPRESENTATION_ITEM()
MEASURE_WITH_UNIT(LENGTH_MEASURE(value),#id)
REPRESENTATION_ITEM('')
);
```
where the reading fails:
- nist_ctc_01_asme1_ap242-e1.stp
- nist_ctc_02_asme1_ap242-e2.stp
- nist_ctc_03_asme1_ap242-e2.stp
- nist_ftc_08_asme1_ap242-e2.stp
- nist_stc_10_asme1_ap242-e2.stp

For others the 3rd parameter is of the expected form
```
#id=LENGTH_MEASURE_WITH_UNIT(LENGTH_MEASURE(value),#id);
```
and the reading of the magnitude succeds:
- nist_ctc_05_asme1_ap242-e1.stp
- nist_ftc_06_asme1_ap242-e2.stp
- nist_ftc_07_asme1_ap242-e2.stp
- nist_ftc_09_asme1_ap242-e1.stp
- nist_ftc_10_asme1_ap242-e2.stp
- nist_stc_06_asme1_ap242-e3.stp
- nist_stc_08_asme1_ap242-e3.stp
- nist_stc_09_asme1_ap242-e3.stp

### Expected Behavior

Tolerance values correspond to the file content in some cases. In others the tolerance values are zero.

### Actual Behavior

Tolerance values should correspond to the file content.

### Sample Code or DRAW Tcl Script

        Handle(XCAFDoc_DimTolTool) gdtTool = XCAFDoc_DocumentTool::DimTolTool(doc->Main());
        TDF_LabelSequence labels;
        gdtTool->GetGeomToleranceLabels(labels);
        for (Standard_Integer idx = labels.Lower(); idx <= labels.Upper(); ++idx)
        {
          TDF_Label label = labels.Value(idx);
          Handle(XCAFDoc_GeomTolerance) tol;
          if (label.FindAttribute(XCAFDoc_GeomTolerance::GetID(), tol)) 
          {
            Handle(XCAFDimTolObjects_GeomToleranceObject) tolObj = tol->GetObject();
            std:::cout << "[Value]: " << tolObj->GetValue() << std::endl;
          }
        }

### Operating System

Linux

### Compiler

GCC

### Bitness

64-bit

### OCCT Version

latest

### Additional Files

for failure: nist_ctc_01_asme1_ap242-e1.stp
for success: nist_ctc_05_asme1_ap242-e1.stp

[nist_ctc_05_asme1_ap242-e1.zip](https://github.com/user-attachments/files/18873595/nist_ctc_05_asme1_ap242-e1.zip)

dpasukhi, 2025-02-19:
The changes are done. need to migrate from https://tracker.dev.opencascade.org/view.php?id=33099

dpasukhi, 2025-02-19:
Changes will be not binary compatible, can't be integrated to 7.9.1.
So, only 8.0.1 dev will be (even can be faster then 7.9.1)