Open-Cascade-SAS/OCCT #387: OCCT fails to build with VTK built with vtkOCCTReader enabled

opoplawski, 2025-02-21:
### Description

I am working on updating the Fedora vtk package to 9.3+.  With 9.3 VTK adds a vtkOCCTReader and builds against OCCT.  When then going to rebuild OCCT/opencacade which uses vtk, we get a build error.

See https://gitlab.kitware.com/vtk/vtk/-/issues/19621 for the same issue filed against vtk.  There initial thought is that this is an OCCT issue.

### Expected Behavior

Successful compilation

### Actual Behavior

```
+ /usr/bin/cmake -S . -B redhat-linux-build -DCMAKE_C_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_Fortran_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_FULL_SBINDIR:PATH=/usr/bin -DCMAKE_INSTALL_SBINDIR:PATH=bin -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DLIB_SUFFIX=64 -DBUILD_SHARED_LIBS:BOOL=ON -GNinja -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUSE_RAPIDJSON=True -DUSE_TBB=True -D3RDPARTY_TBB_LIBRARY_DIR=/usr/lib64 -D3RDPARTY_TBB_INCLUDE_DIR=/usr/include -DUSE_VTK=True -DINSTALL_VTK=False -D3RDPARTY_VTK_LIBRARY_DIR=/usr/lib64 -D3RDPARTY_VTK_INCLUDE_DIR=/usr/include -DBUILD_DOC_Overview:BOOL=OFF -DINSTALL_DIR_LIB=lib64 -DINSTALL_DIR_CMAKE=lib64/cmake/opencascade
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- The C compiler identification is GNU 15.0.1
-- The CXX compiler identification is GNU 15.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Info: TCL is used by OCCT
-- Found Tclsh: /bin/tclsh (found version "9.0")
-- Info: TCL version isn't found
-- Info: TK is used by OCCT
-- Info: Xlib is used by OCCT
-- Info: FreeType is used by OCCT
-- Found Freetype: /usr/lib64/libfreetype.so (found version "2.13.3")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
-- Found OpenGL: /usr/lib64/libOpenGL.so
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5") found components: C
CMake Warning (dev) at /usr/lib64/cmake/vtk/VTK-vtk-module-find-packages.cmake:538 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib64/cmake/vtk/vtk-config.cmake:166 (include)
  adm/cmake/vtk.cmake:63 (find_package)
  CMakeLists.txt:34 (include)
  CMakeLists.txt:625 (OCCT_INCLUDE_CMAKE_FILE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PkgConfig: /usr/bin/pkg-config (found version "2.3.0")
CMake Warning (dev) at /usr/share/cmake/Modules/CMakeFindDependencyMacro.cmake:76 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib64/cmake/libLAS/liblas-config.cmake:18 (find_dependency)
  /usr/lib64/cmake/vtk/VTK-vtk-module-find-packages.cmake:914 (find_package)
  /usr/lib64/cmake/vtk/vtk-config.cmake:166 (include)
  adm/cmake/vtk.cmake:63 (find_package)
  CMakeLists.txt:34 (include)
  CMakeLists.txt:625 (OCCT_INCLUDE_CMAKE_FILE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/lib64/cmake/vtk/VTK-vtk-module-find-packages.cmake:932 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib64/cmake/vtk/vtk-config.cmake:166 (include)
  adm/cmake/vtk.cmake:63 (find_package)
  CMakeLists.txt:34 (include)
  CMakeLists.txt:625 (OCCT_INCLUDE_CMAKE_FILE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/lib64/cmake/vtk/VTK-vtk-module-find-packages.cmake:1402 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib64/cmake/vtk/vtk-config.cmake:166 (include)
  adm/cmake/vtk.cmake:63 (find_package)
  CMakeLists.txt:34 (include)
  CMakeLists.txt:625 (OCCT_INCLUDE_CMAKE_FILE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/lib64/cmake/vtk/VTK-vtk-module-find-packages.cmake:1449 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/lib64/cmake/vtk/vtk-config.cmake:166 (include)
  adm/cmake/vtk.cmake:63 (find_package)
  CMakeLists.txt:34 (include)
  CMakeLists.txt:625 (OCCT_INCLUDE_CMAKE_FILE)
This warning is for project developers.  Use -Wno-dev to suppress it.

VTK version (9.3.1)
-- Info: The directories of 3rdparty headers: 
	/usr/include
	/usr/include/freetype2
-- Info: The directories of 3rdparty libraries: 
	/usr/lib64
-- 
Info: (00:00:00) Start collecting all OCCT header files into /builddir/build/BUILD/opencascade-7.9.0-build/occt-V7_8_1/redhat-linux-build/inc ...
-- Info: (00:00:00) Compare FILES with files in package directories...
-- Info: (00:00:00) Create header-links in inc folder...
-- Info: (00:00:00) Checking headers in inc folder...
-- Info: (00:00:00) End the collecting
CMake Error at CMakeLists.txt:1040 (file):
  file failed to open for writing (Permission denied):

    /usr/bin/custom.sh

CMake Error at adm/cmake/occt_toolkit.cmake:267 (add_library):
  add_library cannot create target "TKernel" because an imported target with
  the same name already exists.
Call Stack (most recent call first):
  CMakeLists.txt:34 (include)
  src/TKernel/CMakeLists.txt:3 (OCCT_INCLUDE_CMAKE_FILE)

CMake Error at adm/cmake/occt_toolkit.cmake:303 (install):
  install TARGETS given target "TKernel" which does not exist.
Call Stack (most recent call first):
  CMakeLists.txt:34 (include)
  src/TKernel/CMakeLists.txt:3 (OCCT_INCLUDE_CMAKE_FILE)

CMake Error at adm/cmake/occt_toolkit.cmake:482 (target_link_libraries):
  Cannot specify link libraries for target "TKernel" which is not built by
  this project.
Call Stack (most recent call first):
  CMakeLists.txt:34 (include)
  src/TKernel/CMakeLists.txt:3 (OCCT_INCLUDE_CMAKE_FILE)

-- Configuring incomplete, errors occurred!
```

### Sample Code or DRAW Tcl Script

N/A

### Operating System

Linux

### Compiler

GCC

### Bitness

64-bit

### OCCT Version

7.9

opoplawski, 2025-02-21:
So I think the problem is that OCCT is importing too many components from VTK.  If I make this change:
```
diff --git a/adm/cmake/vtk.cmake b/adm/cmake/vtk.cmake
index 48650fe7df..8f284b97aa 100644
--- a/adm/cmake/vtk.cmake
+++ b/adm/cmake/vtk.cmake
@@ -2,7 +2,7 @@

 # vcpkg processing
 if (BUILD_USE_VCPKG)
-  find_package(VTK REQUIRED)
+  find_package(VTK REQUIRED COMPONENTS CommonCore)
   set(CSF_VTK VTK::CommonCore)
   set(IS_VTK_9XX 1)
   if (WIN32)
@@ -74,7 +74,7 @@ if (3RDPARTY_VTK_DIR AND EXISTS "${3RDPARTY_VTK_DIR}")
   set (ENV{VTK_DIR} "${3RDPARTY_VTK_DIR}")
 endif()

-find_package(VTK QUIET)
+find_package(VTK COMPONENTS CommonCore QUIET)

 if (3RDPARTY_VTK_DIR AND EXISTS "${3RDPARTY_VTK_DIR}")
   set (ENV{VTK_DIR} ${CACHED_VTK_DIR})
```
Then I don't get the initial import error.  But then I get:
```
VTK version (9.3.1)
-- Info: The directories of 3rdparty headers: 
	/usr/include
	/usr/include/freetype2
-- Info: The directories of 3rdparty libraries: 
	/usr/lib64
-- 
Info: (00:00:00) Start collecting all OCCT header files into /builddir/build/BUILD/opencascade-7.9.0-build/occt-V7_9_0/redhat-linux-build/inc ...
-- Info: (00:00:00) End the collecting
-- Info: Used native memory manager
-- Info: (00:00:00) OCCT toolkits processed
-- Info: (00:00:00) OCCT configuration files prepared
-- Configuring done (2.1s)
CMake Error at /usr/lib64/cmake/vtk/VTK-targets.cmake:674 (set_target_properties):
  The link interface of target "VTK::RenderingUI" contains:

    X11::X11

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/lib64/cmake/vtk/vtk-config.cmake:145 (include)
  adm/cmake/vtk.cmake:77 (find_package)
  CMakeLists.txt:92 (include)
  CMakeLists.txt:919 (OCCT_INCLUDE_CMAKE_FILE)

CMake Error at /usr/lib64/cmake/vtk/VTK-targets.cmake:682 (set_target_properties):
  The link interface of target "VTK::glew" contains:

    GLEW::GLEW

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/lib64/cmake/vtk/vtk-config.cmake:145 (include)
  adm/cmake/vtk.cmake:77 (find_package)
  CMakeLists.txt:92 (include)
  CMakeLists.txt:919 (OCCT_INCLUDE_CMAKE_FILE)

CMake Error at /usr/lib64/cmake/vtk/VTK-targets.cmake:689 (set_target_properties):
  The link interface of target "VTK::RenderingOpenGL2" contains:

    X11::X11

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/lib64/cmake/vtk/vtk-config.cmake:145 (include)
  adm/cmake/vtk.cmake:77 (find_package)
  CMakeLists.txt:92 (include)
  CMakeLists.txt:919 (OCCT_INCLUDE_CMAKE_FILE)
```
I don't know if this is because we're not importing enough components at this point, or if there is a vtk issue.

dpasukhi, 2025-03-03:
@opoplawski please validate that #395 fixed your issue.

closed 2025-03-10 by dpasukhi