Open-Cascade-SAS/OCCT #388: Install error for OpenCASCADEApplicationFrameworkTargets-relwithdebinfo.cmake

opoplawski, 2025-02-21:
### Description

Working on updating the Fedora opencascade package to 7.9.0 and getting an install error.  Looks like an absolute path is getting specified somehow for that target.  Everything else is going into $DESTDIR as expected.

### Expected Behavior

No errors

### Actual Behavior

```
-- Installing: /home/orion/fedora/opencascade/opencascade-7.9.0-build/BUILDROOT/usr/lib64/cmake/opencascade/OpenCASCADEDrawTargets.cmake
-- Installing: /home/orion/fedora/opencascade/opencascade-7.9.0-build/BUILDROOT/usr/lib64/cmake/opencascade/OpenCASCADEDrawTargets-relwithdebinfo.cmake
CMake Error at redhat-linux-build/cmake_install.cmake:7954 (file):
  file failed to open for writing (Permission denied):

    /usr/lib64/cmake/opencascade/OpenCASCADEApplicationFrameworkTargets-relwithdebinfo.cmake
```

### Sample Code or DRAW Tcl Script

N/A

### Operating System

Linux

### Compiler

GCC

### Bitness

64-bit

### OCCT Version

latest

dpasukhi, 2025-02-27:
@opoplawski could you please share how do you call cmake?
I can't reproduce for some reason.

dpasukhi, 2025-02-27:
Will it help?
https://github.com/Open-Cascade-SAS/OCCT/pull/398

opoplawski, 2025-02-28:
cmake invocation:
```
+ /usr/bin/cmake -S . -B redhat-linux-build -DCMAKE_C_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_Fortran_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_FULL_SBINDIR:PATH=/usr/bin -DCMAKE_INSTALL_SBINDIR:PATH=bin -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DLIB_SUFFIX=64 -DBUILD_SHARED_LIBS:BOOL=ON -GNinja -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUSE_RAPIDJSON=True -DUSE_TBB=True -D3RDPARTY_TBB_LIBRARY_DIR=/usr/lib64 -D3RDPARTY_TBB_INCLUDE_DIR=/usr/include -DUSE_VTK=True -DINSTALL_VTK=False -D3RDPARTY_VTK_LIBRARY_DIR=/usr/lib64 -D3RDPARTY_VTK_INCLUDE_DIR=/usr/include -DBUILD_DOC_Overview:BOOL=OFF -DINSTALL_DIR_LIB=lib64 -DINSTALL_DIR_CMAKE=lib64/cmake/opencascade
```

But your pull request seems to fix the issue.  Thanks!

opoplawski, 2025-02-28:
Install is:
```
+ DESTDIR=/builddir/build/BUILD/opencascade-7.9.0-build/BUILDROOT
+ /usr/bin/cmake --install redhat-linux-build
```

closed 2025-03-03 by dpasukhi