Open-Cascade-SAS/OCCT #430: The model has defects when converting from step format to wavefront format

xiaomx32, 2025-03-12:
### Description

- create topological shapes
- generate mesh data using `BRepMesh_IncrementalMesh`
- using `BRep_Tool::Triangulation` extracts data
- manually export as wavefront model (vertices and faces)

Taking the simplest 3D surface (sphere) as an example, it is confusing that the radius of the sphere and the `theLinReflection` parameter of `BRepMesh_IncrementalMesh` affect the appearance of the final obj model.

Tested versions 7.8 and 7.9, both have this problem.

### Expected Behavior

Export the correct wavefront model，the vertices are evenly distributed with the faces, without any strange defects

### Actual Behavior

Using Blender to view the final exported obj file, the appearance is as follows.
From left to right, the radius of the ball is 0.5, 4.5, 13.5 (for ease of display, I manually scaled them to a uniform size in Blender, which does not affect the results at all).
A defect appeared along the seam edge of a sphere with smaller radius.

[image]

[image]

### Sample Code or DRAW Tcl Script

The minimum example code is as follows:

```
#include <opencascade/BRepPrimAPI_MakeSphere.hxx>
#include <opencascade/BRepMesh_IncrementalMesh.hxx>
#include <opencascade/TopExp_Explorer.hxx>
#include <opencascade/BRep_Tool.hxx>
#include <opencascade/TopoDS.hxx>

#include <iostream>
#include <string>
#include <numbers>

int main() {
    const TopoDS_Shape sphere = BRepPrimAPI_MakeSphere(3.0, 360.0 * std::numbers::pi_v<double> / 180.0).Shape();

    std::ofstream obj_file("step2obj_test.obj");
    if (!obj_file.is_open()) { std::cerr << "Failed to open file" << std::endl; }

    BRepMesh_IncrementalMesh(sphere, 0.01);
    int vertex_index { 0 };
    for (TopExp_Explorer explorer(sphere, TopAbs_FACE); explorer.More(); explorer.Next()) {
        TopoDS_Face face = TopoDS::Face(explorer.Current());
        TopLoc_Location loc;
        Handle(Poly_Triangulation) triangulation = BRep_Tool::Triangulation(face, loc);
        if (triangulation.IsNull()) { std::cerr << "triangulation is null" << std::endl; }
        for (int i = 1; i <= triangulation->NbNodes(); ++i) {
            const gp_Pnt& pnt = triangulation->Node(i);
            obj_file << "v " << pnt.X() << " " << pnt.Y() << " " << pnt.Z() << '\n';
        }
        for (int i = 1; i <= triangulation->NbTriangles(); ++i) {
            const Poly_Triangle& triangle = triangulation->Triangle(i);
            int n1 { 0 };
            int n2 { 0 };
            int n3 { 0 };
            triangle.Get(n1, n2, n3);
            obj_file << "f " << n1 + vertex_index << "// " << n2 + vertex_index << "// " << n3 + vertex_index << "//" << '\n';
        }
        vertex_index += triangulation->NbNodes();
    }

    std::cout << "conversion complete." << std::endl;

    return 0;
}
```

### Operating System

Linux

### Compiler

GCC

### Bitness

64-bit

### OCCT Version

7.8

xiaomx32, 2025-03-12:
Recently, after extensive testing, it was found that the problem almost always occurs in the triangulation of surfaces.
By the way, `BRep_Tool` is used here: triangulation to obtain vertex data and write it to obj is not intentional, as `RWObj_CafWriter` has the same issue. Therefore, I tend to believe that the problem lies with `BRepMesh_IncrementalMesh`.

dpasukhi, 2025-03-12:
Dear @xiaomx32 
The issue is related to the parameters that you are using.
0.1 deflection for the 0.5 radius is a not applicable parameter. Using hardcore parameter is not best result.
ExpressMesh offers option to find the comfortable parameter, but in my case i can share sample, which I use in some projects where needs to work with small and huge elements on the same time.

```
  Handle(XCAFDoc_ShapeTool) aShapeTool = XCAFDoc_DocumentTool::ShapeTool(theDoc->Main());
  TopoDS_Shape aSingleShape = aShapeTool->GetOneShape(aFreeShapes);
```
```

#include <BRepBndLib.hxx>
#include <BRepBuilderAPI_Copy.hxx>
#include <BRepBuilderAPI_GTransform.hxx>
#include <BRepBuilderAPI_MakeEdge.hxx>
#include <BRepBuilderAPI_MakeFace.hxx>
#include <BRepBuilderAPI_MakeWire.hxx>
#include <BRepBuilderAPI_Transform.hxx>
#include <BRepMesh_IncrementalMesh.hxx>
#include <BRepPrimAPI_MakeBox.hxx>
#include <BRepPrimAPI_MakeCone.hxx>
#include <BRepPrimAPI_MakeCylinder.hxx>
#include <BRepPrimAPI_MakeSphere.hxx>
#include <BRepPrimAPI_MakeTorus.hxx>
#include <Bnd_Box.hxx>
#include <TopExp_Explorer.hxx>

namespace
{
  static const double THE_LINEAR_DEFLECTION  = 5.e-3;
  static const double THE_ANGULAR_DEFLECTION = 0.5;
  static const bool   THE_IS_PARALLEL        = false;
}

bool GTests::Utils::Tessellate(const TopoDS_Shape& theShape)
{
  bool       anIsDone      = false;
  const auto aShapeProcess = [&](const TopoDS_Shape& theSubShape)
  {
    Bnd_Box aBounds;
    BRepBndLib::Add(theSubShape, aBounds);

    IMeshTools_Parameters aParameters;
    const gp_XYZ          aMin           = aBounds.CornerMin().XYZ();
    const gp_XYZ          aMax           = aBounds.CornerMax().XYZ();
    const double          aDeflection    = std::max((aMax - aMin).Modulus() * THE_LINEAR_DEFLECTION, THE_LINEAR_DEFLECTION);
    aParameters.Deflection               = aDeflection;
    aParameters.MinSize                  = aDeflection * 2;
    aParameters.Angle                    = THE_ANGULAR_DEFLECTION;
    aParameters.InParallel               = THE_IS_PARALLEL;
    aParameters.ControlSurfaceDeflection = true;
    aParameters.EnableControlSurfaceDeflectionAllSurfaces = true;
    const BRepMesh_IncrementalMesh aTessellated(theSubShape, aParameters);
    anIsDone = anIsDone || aTessellated.IsDone();
  };
  for (TopExp_Explorer aSolidIter(theShape, TopAbs_SOLID); aSolidIter.More(); aSolidIter.Next())
  {
    aShapeProcess(aSolidIter.Current());
  }
  for (TopExp_Explorer aFaceIter(theShape, TopAbs_FACE, TopAbs_SOLID); aFaceIter.More(); aFaceIter.Next())
  {
    aShapeProcess(aFaceIter.Current());
  }
  for (TopExp_Explorer anEdgeIter(theShape, TopAbs_EDGE, TopAbs_FACE); anEdgeIter.More(); anEdgeIter.Next())
  {
    aShapeProcess(anEdgeIter.Current());
  }
  return anIsDone;
}
```

dpasukhi, 2025-03-12:
@xiaomx32 the code sample is help with your case?

xiaomx32, 2025-03-13:
You are right, there is a subtle relationship between the size of an object and the parameters of `BRepMesh_IncrementalMesh`.
Thank you for your code sample, but from the results, it doesn't seem good as the vertices after meshing are even more chaotic.
Or is it true that current algorithms do not support objects that are too small in size (for example, 0.5mm-5mm)?

dpasukhi, 2025-03-13:
Theoretically it is supported.
But need to specify linear and angular deflection with low parameters.
As for singularity, the angular is used, you can make it 0.1 for example instead of 0.5 in my sample.

xiaomx32, 2025-03-13:
The values of `Deflection` and `Angle` have a significant impact on the results, and it is too difficult to adjust these two parameters to a suitable value to accommodate models of different sizes.

xiaomx32, 2025-03-13:
Yes, this may be the best approach currently available.
It is worth mentioning that if `TopoDS_Shape` is composed of multiple shape, it should be based on the bounding box of the smallest object.

dpasukhi, 2025-03-13:
TopExp_Explorer extract highest level.
It means it split compound into solids.
You can specify your own splitting.

xiaomx32, 2025-03-13:
Thank you, thank you for your guidance.