Open-Cascade-SAS/OCCT #455: minGW64 failed to build and link OpenCASCADE libTKBool.dll

aruig666, 2025-03-20:
### Description

I want to rebuild OpenCASCADE with minGW64. But there were some problems. This is all the configuration of cmake:

[image]
[image]
[image]
[image]

First,when I compile, the program often stalls and needs to be re-run to continue compiling (I put up with it. I ran it over and over and I got 27% of the way there)

But,I still got an error when compiling (or should I say linking) libTKBool.dll.

[image]

The error appears to have automatically switched to the 32-bit ld.exe.
gcc --version
g++ --version
gdb --version
[image]

### Expected Behavior

A one-time compilation is expected

### Actual Behavior

 the program often stalls and needs to be re-run to continue compiling

### Sample Code or DRAW Tcl Script

cmake --build .

### Operating System

Windows

### Compiler

GCC

### Bitness

64-bit

### OCCT Version

7.7

dpasukhi, 2025-03-21:
@aruig666 please check the latest release.
Mingw build is fixed and weekly tested.
See:
https://github.com/Open-Cascade-SAS/OCCT/blob/master/.github/workflows/build-multiconfig-mingw.yml

aruig666, 2025-03-21:
@dpasukhi  Thank you very much for your answer. I tried it, but there are still some problems.
    - name: Configure basic  (line 75 in build-multiconfig-mingw.yml)
I followed this configuration by adding these sentences to the beginning of cmakelist.
`
set (CMAKE_C_COMPILER "x86_64-w64-mingw32-gcc")
set (CMAKE_CXX_COMPILER "x86_64-w64-mingw32-g++")
set (CMAKE_CXX_COMPILER "x86_64-w64-mingw32-g++")
set (CMAKE_PREFIX_PATH "C:/msys64/mingw64")
set (CMAKE_BUILD_TYPE "Release")
`
>  cmake -G "Ninja" ..
Compile log as follows：
`
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/x86_64-w64-mingw32-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/mingw64/bin/x86_64-w64-mingw32-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Info: TCL is used by OCCT
-- Info: TK is used by OCCT
-- Info: FreeType is used by OCCT
-- Info: TKIVtk and TKIVtkDraw toolkits excluded due to VTK usage is disabled
-- Found Tclsh: C:/msys64/mingw64/bin/tclsh.exe (found version "8.6")
-- Info: TCL version isn't found
-- Found Freetype: C:/msys64/mingw64/lib/libfreetype.a (found version "2.13.3")
-- Info: The directories of 3rdparty headers:
        C:/msys64/mingw64/include
        C:/msys64/mingw64/include/freetype2
-- Info: The directories of 3rdparty libraries:
        C:/msys64/mingw64/lib
        C:/msys64/mingw64/bin
CMake Warning at CMakeLists.txt:1005 (message):
  Could not find DLLs of used third-party products: 3RDPARTY_TCL_DLL_DIR
--Info: (02:33:00) Start collecting all OCCT header files into D:/SourceFiles/occ_work/v7_9/OCCT-7_9_0/build/inc ...
-- Info: (02:33:10) End the collecting
-- Info: Used native memory manager
-- Info: (02:33:10) OCCT toolkits processed
-- Info: (02:33:10) OCCT configuration files prepared
-- Configuring done (15.3s)
-- Generating done (2.4s)
-- Build files have been written to: D:/SourceFiles/occ_work/v7_9/OCCT-7_9_0/build
`
Looks like a lot of dependencies can't be found.
I checked it out with the cmake gui:
[image]
There is no ./include ./lib subfolder under C:\msys64\mingw64\include\tcl8.6

dpasukhi, 2025-03-22:
Dear @aruig666 could you please use CMake from MINGW terminal or use 3rd-party from the  package?

aruig666, 2025-03-22:
Dear @dpasukhi 
This is my entire cmake configuration
[image]

[image]

[image]

[image]

`OCCT-7_9_0\build>cmake --build .` 
Error message：
[image]

gcc path
[image]

dpasukhi, 2025-03-22:
Unfortunately, I can't reproduce that with Msys2, which i'm using.
If MinGW have bash terminal, please try to use from here.
Additionally, could you check Clang compiler and C++11 version, just in case.
_
At the moment, I don't have any ideas.

dpasukhi, 2025-04-02:
Was trying to reproduce on others PC with Msys2.
Not reproducible on latest version.
Assume the problem with local environmental.

closed 2025-04-02 by dpasukhi