Open-Cascade-SAS/OCCT #469: Bounding box is void when using

rajendrasing-pardeshi, 2025-03-30:
### Description

I am getting different behavior for two APIs- 
1. BRepBndLib::AddClose
I get folloing bbox - 
<HTML>
<head>
<title>Document</title></head>
<body>

  | Name | Value | Type
-- | -- | -- | --
  | Xmin | -10.413873542049933 | double
  | Xmax | 21.686673542049927 | double
  | Ymin | -16.927663351829082 | double
  | Ymax | 23.618863351829077 | double
  | Zmin | -135.72389864127800 | double
  | Zmax | 59.395000000000003 | double
  | Gap | 0.0000000000000000 | double
  | Flags | 0 | int

</body>
</HTML>

2. BRepBndLib::AddOptimal
I get exception with message - 
<HTML>
<head>
<title>Document</title></head>
<body>

  | Name | Value | Type
-- | -- | -- | --
▶ | Message | 0x000001dc729e91b4 "Bnd_Box is void" | char[1]

</body>
</HTML>

I am attaching shape. 
Is this fixed in latest version?
Which is a reliable method to get bounding box?

### Expected Behavior

Should return bounding box and not crash.

### Actual Behavior

Exception is thrown.

### Sample Code or DRAW Tcl Script

Standard_Real xMin, yMin, zMin, xMax, yMax, zMax;
Bnd_Box bndBox;
try
{
				BRepBndLib::AddOptimal(shape, bndBox);
}
catch (std::exception& ex)
{
				ex;
				return;
}
catch (Standard_Failure const& fail)
{
				fail;
				return;
}

### Operating System

Windows

### Compiler

MSVC

### Bitness

64-bit

### OCCT Version

7.7

### Additional Files

[cutCompound2.zip](https://github.com/user-attachments/files/19525260/cutCompound2.zip)

dpasukhi, 2025-03-30:
Fix is going:
 
di << axmin << " " << aymin << " " << azmin << " " << axmax << " " << aymax << " " << azmax;
-10.383446753040507 -12.671863137694107 -135.723898641278 21.647222384757271 19.363063137694109 59.414846753040507

rajendrasing-pardeshi, 2025-03-31:
Thank you @dpasukhi for fixing.
I am using version 7.7. Which version is this fixed in? Will the fix be available in 7.7?

dpasukhi, 2025-03-31:
No.
Patching old version is not available for community requests.
It will be part of master

closed 2025-03-31 by dpasukhi