Open-Cascade-SAS/OCCT #553: ElCLib::AdjustPeriodic() failed to add a period

xuzhongxing, 2025-05-20:
### Description

In the function ElCLib::AdjustPeriodic(), I got the following values:
```
U1: 0.98620381059140871
U2: 0.98620381138328739
Preci: 2.3342157359074867e-10
```
Thus, U2 - U1 < Preci is not satisfied. And U2 += period is not executed, resulting in U1 == U2, which is wrong.

Preci is obtained by GeomAdaptor_Surface::UResolution(const Standard_Real R3d). The surface is a bspline surface.

In Geom_BSplineSurface::Resolution(), umaxderivinv is 2e-3, and Tolereance3D is 1e-7.

It seems that I have nowhere to adjust these parameters.

Are there parameters that I can set to work around this default behavior? Or, should some logic be changed to avoid this?

### Expected Behavior

U2-U1<Preci should be satisfied to add a period to get a correct U2.

### Actual Behavior

U2 - U1 < Preci is not satisfied. And U2 += period is not executed, resulting in U1 == U2, which is wrong.

### Sample Code or DRAW Tcl Script

This occurs during the computation of Adaptor3d_IsoCurve::Load(). I cannot extract a test case easily for this.

### Operating System

Windows

### Compiler

MSVC

### Bitness

64-bit

### OCCT Version

7.9

xuzhongxing, 2025-05-20:
I think there is some logic problem in ElCLib::AdjustPeriodic(). If `(U2-U1)` is a little more than `(period + Preci)`, then U2 would be decreased by one period and never compensated by `(u2+= period)` in line 117.

If the purpose of adjusting U2 is to set U2 exactly to U1 + period, then the check on line 116 `if (U2 - U1 < Preci)` should be removed.

xuzhongxing, 2025-05-20:
The patch gives a slightly different semantic for the function.

Under old logic:

For U1: 0, U2: 1.2, the function gives: U1:0, U2:0.2.
For U1: 0, U2: 1.000001, the function gives: U1:0, U2: 0.000001

Under new logic:

For U1: 0, U2: 1.2, the function gives: U1: 0, U2: 1
For U1: 0, U2: 1.000001, the function gives: U1: 0, U2: 1

I think the new logic makes more sense.

dpasukhi, 2025-05-20:
Please share the code and file to reproduce.
And share the error, what's impact of current behaviour.

xuzhongxing, 2025-05-20:
```
#include "TopoDS_Shape.hxx"
#include "TopoDS.hxx"
#include "BRepTools.hxx"
#include "BRep_Builder.hxx"
#include "Adaptor3d_IsoCurve.hxx"
#include "BRepAdaptor_Surface.hxx"
#include <fstream>
#include <filesystem>

int main() {
    std::filesystem::path input("D:/occttest/testcase.brep");
    std::ifstream ifs(input, std::ios::binary | std::ios::in);
    TopoDS_Shape shape;
    BRep_Builder builder;
    BRepTools::Read(shape, ifs, builder);

    TopoDS_Face face = TopoDS::Face(shape);

    Handle(BRepAdaptor_Surface) aSurf(new BRepAdaptor_Surface(face));

    Adaptor3d_IsoCurve targetCurve(aSurf, GeomAbs_IsoV, 0);

    std::cerr << "param: " << targetCurve.FirstParameter() << " " << targetCurve.LastParameter() << "\n";

    return 0;
}

[testcase.zip](https://github.com/user-attachments/files/20327782/testcase.zip)

```