Open-Cascade-SAS/OCCT #633: VCPKG : include error, on IntCurve_IntConicCurveGen.lxx using vcpkg

mattepella, 2025-07-21:
### Description

Hello, i integrated opencascade in my visual studio project using vcpkg, and if i specify the include directory like "path/to/include/opencascade/", i have no error while compiling.
although if I use vcpkg i would like to not specify any include directories manually.
The specifc error i got is coming from this class: IntCurve_IntConicCurveGen.lxx

the error:
IntCurve_IntConicCurveGen.lxx(17,10): error C1083: not possible to open include file: 'HLRBRep_TheIntersectorOfTheIntConicCurveOfCInter.hxx': No such file or directory

If i look at the error deeply i see this line on a .lxx file that seems to cause problems:

#include IntCurve_TheIntersector_hxx

#include "IntRes2d_Domain.hxx"

#include "gp_Lin2d.hxx"
#include "gp_Circ2d.hxx"
#include "gp_Elips2d.hxx"
#include "gp_Hypr2d.hxx"
#include "gp_Parab2d.hxx"

### Expected Behavior

i shouldn't get the error, and shouldn't set include path in visual studio

### Actual Behavior

the project should compile without giving me the error (it will work if i specify the ../opencascade/ directory)

### Sample Code or DRAW Tcl Script

the error:
IntCurve_IntConicCurveGen.lxx(17,10): error C1083: not possible to open include file: 'HLRBRep_TheIntersectorOfTheIntConicCurveOfCInter.hxx': No such file or directory

#include IntCurve_TheIntersector_hxx

#include "IntRes2d_Domain.hxx"

#include "gp_Lin2d.hxx"
#include "gp_Circ2d.hxx"
#include "gp_Elips2d.hxx"
#include "gp_Hypr2d.hxx"
#include "gp_Parab2d.hxx"

### Operating System

Windows

### Compiler

MSVC

### Bitness

32-bit

### OCCT Version

7.9

dpasukhi, 2025-07-22:
Sorry, do you expect to have "" instead of <>?

mattepella, 2025-07-22:
No the issue is that project is not compiling, and ure right if i take a look at the source code u have on github u actually have it correct, idk why tho i have the files like that.

dpasukhi, 2025-07-22:
The problem on vcpkg port, not OCCT side.
Please create an issue for VCPKG repo.

The code which making that. replacement: https://github.com/microsoft/vcpkg/blob/ebc2e14093db29fbd8277c7ec8e4eb81a222fdc3/ports/opencascade/portfile.cmake#L58