Open-Cascade-SAS/OCCT #654: Consider providing an official C API

conorbergin, 2025-07-30:
### Description

Creating language bindings for OCCT usually requires writing an intermediate C interface, most languages have no reliable way to bind to C++, and C APIS are an accepted lingua franca. Examples of projects that do this are [waterfall-cad](https://github.com/joe-warren/opencascade-hs) and [opencascade-rs](https://github.com/bschwind/opencascade-rs), I'm sure there are others I am unaware of.

Gmsh, which uses OCCT for modelling, has an auto-generated C API, so you can use OCCT as a C library today through Gmsh, although it  has quite a different API.

### Use Case

Creating language bindings, calling OCCT from languages that support calling C directly.

### Benefits

Make OCCT easy to use from other languages.

dpasukhi, 2025-07-30:
Hello. Thank you for your feature request. OCCT is OOP library and just create and wrap into C is not easiest task.
We have our C# and Java wrapper. Also there are a few open source wrappers which wrapped existed c++ classes as is.
At the moment, OCCT have no wrappers in roadmap.
But there a small interesting news, I'm working on own pet-project, which offer simplified version of OCCT and provides JS/TS, C# and python interfaces (others maybe will be added later). That project is a new headers without access to OCCT headers. As a result, logic is simpler, but performance is lower.
It is not C, but have really simplified headers and logic, which also can be easily wrapped by C.
I'm not sure when I will publish it, or move it into occt repo itself. But I assume first version will be available that year.
Best regards, Dmitrii.

conorbergin, 2025-09-13:
Hi, I've just seen your update, that looks great.  Looking further into SWIG it appears they do support [C++ -> C](https://www.swig.org/Doc4.3/C.html) as of last year, I look forward to testing it out.